osiweb/unified_retro_keyboard #37: Knight and SAIL keyboards

larsbrinkhoff, 2021-12-14:
Hello,

Would it be possible to adapt this to make Knight and SAIL keyboards?

larsbrinkhoff, 2021-12-14:
The Knight keyboard was used with the MIT AI lab PDP-10 running ITS, and later also for the first Lisp machines before Kulp designed the infamous space-cadet keyboard.

[image]

larsbrinkhoff, 2021-12-14:
The SAIL keyboard was designed by Les Earnest for use with the Stanford AI Lab PDP-10.  It predates the Knight keyboard and has pretty much the same features.

[image]

dfnr2, 2021-12-14:
Yes, it would be possible to add these keyboards. If you are interested in seeing these keyboards replicated in the future, please provide information on the physical interface as well as the data protocol.  It wouldn't hurt if someone contributed a PCB layout compatible with the 40-pin encoder interface.  The classic layout could serve as a starting point.

The keycaps could be made, but the custom symbols might be a substantial up front charge from most manufacturers.  Resin-printing the 3D keycaps may be reasonable, though.

larsbrinkhoff, 2021-12-15:
Thank you.

I think the physical interface or protocol would not be useful since the computers on the other end are long gone, and now only exists as emulators.  So an USB or PS/2 interface would make more sense.

In what ballpark would the cost of the custom keycaps be?  I imagine it could be amortized over a small but dedicated community of PDP-10 enthusiasts.

dfnr2, 2021-12-15:
All this is doable.  PS/2 emulation  would be straightforward to add, and I believe a soft USB/HID emulation could be done, or an arch-* module could be added for a chip with USB engine and PHY.  

I have inquired whether the greek & match symbols are available or if they require a setup fee.  The fee is $50 per new symbol.  It looks like either keyboard would need around 30 new symbols, so $1500, only one time.  An order of 10 keyboards would probably be in the range of $100-$120, so let's say the first order of 10 keyboards would probalby run about $250/set of keys.  Subsequent orders would be in the $100-$120 range.  Bigger orders would be less per set.

Of course, another route would be to create a 3D keycap model, then emboss the legends into the keys in CAD, and 3D print the whole thing on a resin printer using colored resin.  If anyone wants to contribute some keycap models, I would welcome them.

larsbrinkhoff, 2021-12-16:
Pricey, but I think many will think it's worth it.  I'll ask around to gauge the interest.

I don't think I could pull off the 3D print, but I'll keep it in mind and see if someone else can.

larsbrinkhoff, 2021-12-17:
Seems there could be a first batch of 40-50 keyboards, so looks like it's a feasible project.

So next step would be to design the PCBs, correct?  Is there anything else I need to do at this stage, or more information to provide to you?

larsbrinkhoff, 2021-12-17:
For reference, @MMcM's Knight keyboard:
[image]

larsbrinkhoff, 2021-12-17:
Various photos of the SAIL keyboard can be found here: 
https://deskthority.net/wiki/SAIL_keyboard

dfnr2, 2021-12-19:
I think the big variable here is the keycaps.  The next step is to get the keycap order placed.  You should start with the spreadsheet for the classic keycaps, and extend to the Knight or SAIL layout.  My suggestion would be to go with the Knight layout, since it seems closer to the modern layouts.  Either way, I think that should be the first step.

Once an order is placed, there is likely an 18 month lead time (as of Dec. 2021), so there is plenty of time to get the PCB and keymaps ready.  I would start from the classic keyboard layout, and extend to the selected layout.

Finally, the last step is to add the layout to the firmware.

I will be happy to assist with any of those steps.  I would prefer someone take the lead on getting a keycap order together.  I will be happy to review the spreadsheet.  I will also work with any volunteer who wants to make the PCB design.   If we have keycaps and PCBs on order, then I will be happy to work with someone on the keymap, or make the keymap myself.

larsbrinkhoff, 2021-12-20:
I posted a pull request with a spreadsheet here: #38

dfnr2, 2021-12-20:
I've merged the pull request.  It would be a good idea to have a couple of other experts look it over.  I will also look it over myself.   Is there a list somewhere where you will be discussing this project with other PDP-10'ers?  I would like to join, if only to listen in.

larsbrinkhoff, 2021-12-20:
Great, thanks!  When there's time, I'd like to hear your comments expecially about font sizes.  Is there a way to see some kind of sample from the manufacturer?

There is no longer any one definitive place for the PDP-10 community.  For now, the most interest seems to be from the PiDP-10 group: https://groups.google.com/g/pidp-10  
and the ITS mailing list: http://its.victor.se/mailman/listinfo/its-hackers_its.victor.se

larsbrinkhoff, 2021-12-21:
This is in stock again!  https://novelkeys.com/products/sa-sail  
Maybe it can be of use here.

larsbrinkhoff, 2021-12-21:
Sorry, false alarm.  Almost all options are sold out.

larsbrinkhoff, 2021-12-21:
Some comments from @aap and @ams:

- zero with slash is sort weird, its just a zero
- that one is called circle-cross
- the 0 and Z i suppose you just want to note that they are typeset using a slash, and a line over it ... but sorta confundeling, the way 0 is rendered looks spectacularily weird
- [Might use a Scandinavian Ø instead] or the empty set symbol
- " should be straight

larsbrinkhoff, 2021-12-21:
@dabridgham points out that the spreadsheet specifies row 3 profile for all keys.  It's mostly because I copied another file from here, and I haven't checked what the Knight keyboard looks like.

Another consideration is the awkward placement of the CTRL and META keys.  Should they remain in place, be moved, or both (two different layouts)?

larsbrinkhoff, 2021-12-21:
The available profiles can be found here:  
https://pimpmykeyboard.com/key-cap-family-specs/

I can see why the "classic keyboard" in this repository chose SA row 3.  It might also be the closest match for the Knight keyboard.  It could be that once the up front charge has been paid, people can order their own keycap set with any profile.

dfnr2, 2021-12-21:
Row 3 for all keys is intentional.  That way, keys can be reconfigured from one row to another freely.  Moving CTRL and META keys is a prime example.  Some might want the original position, and others might want to modify to place some keys in a more contemporary position.  If the keycaps are all row 3, it's pretty easy to support multiple configurations.

larsbrinkhoff, 2021-12-22:
More feedback from @dabridgham:
- Do we need to consider mount styles?  https://pimpmykeyboard.com/template/images/MountStyles.pdf
- Design the PCB with extra key positions, to be flexible with regards to key placement.

@dfnr2 wrote:
> IMO, the flexibility is worth it

Yes, I can see this is a good idea!   Regarding the profiles, I have asked for photos.  It's not easy because I only know of one person that has a keyboard.

larsbrinkhoff, 2021-12-22:
SAIL keyboard profile... hard to say from this angle, but looks like all keys have the same flat profile.  I emailed Les Earnest.
[image]

larsbrinkhoff, 2021-12-22:
The related but distinct "space-cadet" or Kulp keyboard does have varying height and slant:
[image]

larsbrinkhoff, 2021-12-22:
https://theworld.com/~jdostale/kbd/Knight.html
> The Backnext key was used for things like terminal linking. Because of the way the metal was positioned, this key makes a different thunk from all the other keys.

larsbrinkhoff, 2021-12-22:
@quadibloc has these layout renditions on his page http://www.quadibloc.com/comp/kyb01.htm and points out this detail about the SAIL keyboard:
> instead of the four rows of the main typing area being staggered by half a key, a quarter of a key, and half a key, as one goes down from the top, each row actually was staggered to the right by half a key.

[image]  
[image]

dabridgham, 2021-12-22:
To clarify, the extra key positions I was talking about had to do with the positions of the CTRL and META keys.  They seem awfully wide on the Knight TV keyboard; it looks like it would be a stretch to type M-g one-handed.  I've never used a Knight TV though, so perhaps I'm wrong about that. I do notice that the CADR keyboards (which I have used and liked) have them in closer.

Lars and I were talking about doing two keyboard PCBs when it occurred to me that there's plenty of room inside of the CTRL keys for another key.  So, I thought, why not just design the PCB with key positions there and each person can populate the keys how they prefer, either like the original or in a little closer if that turns out to be easier to use.

dfnr2, 2021-12-22:
Definitely a good idea to build flexibility in to the main keyboard where you can.  If you can keep keycap sizes the same between layouts, even better, as that cuts down on the size of the keycap order.  

I think that closely replicating the physical details of a keyboard is a great goal, especially for someone trying to repair a vintage machine, in need of a close replica.  But there is also value in making certain compromises that improve the flexibility and usability of the keyboard, or make it more likely to be made, such as using the symmetrical profile (row 3) to allow reuse of keys between different layouts and even different keyboards, choosing colors that look good with multiple keyboards, if not perfect for any, and at least providing alternate key positions where the originals were awful.  I

Personally, I'm all about recreating the experience, but less a stickler for the most minute details, if it means I can make ordering keycaps more practical.  I think that the balance might be tilted toward perfection if we had a great set of 3D keycap models and a tool/script for embossing legends on them in 3D.

larsbrinkhoff, 2021-12-22:
Tom Knight still has some keyboard(s?).  He says he would like Bluetooth.

larsbrinkhoff, 2021-12-23:
Going into this, I didn't plan to make a hyper detailed replica.  So e.g. the exact font on the keycaps doesn't have to be a 100% match.  That said, if we can get good close-up photos, maybe I'll change my mind about that.

> that cuts down on the size of the keycap order.

This brings up a question.  Do we need to place a single order for all keycap sets, for say, 50 keyboards?  It would be good if people could place their own customized order once the initial front up cost has been paid.

larsbrinkhoff, 2021-12-24:
MMcM provided these side photos.  
[image]
[image]

dabridgham, 2021-12-24:
Well that's clear enough then.  No sculpturing on the keys.

As for Bluetooth, I see these little Bluetooth modules that look like they should be pretty easy to use.  I've never played with any though so I don't know first-hand.  Probably the hardest part there would be powering it.  The expectation with wireless is that the keyboard would have a battery of some sort so then you need a charging circuit.  Or maybe just use removable AAs or an 18650 and an external charger.  Perhaps just put a Bluetooth module connection on the USB interface board and leave the question of powering the whole thing to the builder or to a future and separate project.

larsbrinkhoff, 2021-12-29:
I'm reaching out to these manufacturers:
- [Signature Plastics](https://pimpmykeyboard.com/)
- [Drop](https://drop.com/)
- [WASD Keyboards](https://www.wasdkeyboards.com/)

larsbrinkhoff, 2021-12-29:
This renders keyboards layouts very nicely: https://kle-render.herokuapp.com/

[image]
[image]

dabridgham, 2021-12-29:
Is the purpose of that keyboard editor only to produce renders or is there also a path to going from the design to a set of keycaps or even an actual keyboard?  I notice you can specify mount and switches which suggest to me something more but I'm not finding it.

larsbrinkhoff, 2021-12-29:
I have used this: http://www.keyboard-layout-editor.com/  From which I got the pictures at the top of this issue.

It stores its data as a GitHub gist in JSON format.  Besides "kle-render" mentioned above, rumor has it there's some number of tools that can use the data.  Such as making PCB layouts, but I haven't investigated this.

larsbrinkhoff, 2021-12-29:
Here is a list of "KLE" tools: https://www.keyboard-design.com/tools.html

larsbrinkhoff, 2021-12-30:
The text font seems to be the same as other similar vintage Micro Switch keyboards.  Many samples can be found online.  For example:  
[image]

larsbrinkhoff, 2021-12-30:
The Blender plugin works fine: https://github.com/Kirpal/Keyboard-Layout-Editor-for-Blender

[image]
[image]

larsbrinkhoff, 2021-12-31:
Comparing the SAIL and Knight keycap sets.  They share 21 keycaps if we disregard color differences.

[image]

larsbrinkhoff, 2021-12-31:
@dfnr2, is this an option?

https://www.reddit.com/r/cableporn/comments/atfbg4/this_is_what_a_mechanical_keyboard_without_a_pcb/

larsbrinkhoff, 2021-12-31:
Tom Knight informs us on Twitter that the awkward positions of KEYS and CTRL were due to a mistake.

dabridgham, 2022-01-01:
Interesting history from TK.  He also suggested moving the parens to lower-case like they are on both the SAIL and Space Cadet keyboards.  I'm good with either/both modifications but also realize that once we start changing things, we run the risk of a protracted "discussion" while we figure it out.  There's always the tension of how much to recreate exactly and how much to improve.  I figured the idea of just putting pads on the keyboard PCB for another key inboard of the two CTRL keys was a pretty reasonable compromise, allowing the builder to do it either way (or even populating all of the positions and adding HYPER or SUPER).

Speaking of Space Cadet keyboards, I noticed you commented on Twitter that you get a lot of requests for those.  I bet printing on the front of the keycaps would be an issue and probably expensive.  I'd put my hand up for those too, if possible/practical.

dabridgham, 2022-01-01:
Two comments about the keyboard circuit board.  First, the USB and PS/2 protocols as well as dfnr2's 40-pin connector all support LEDs (at least three) so I'd suggest, if there's room, that you put three LED positions on the PCB.  These original keyboards didn't have LEDs of course but it'd be trivial to just leave them off when building the board if you didn't want them.

Second, regarding that 40-pin connector, I see that the other keyboards defined here put that connector near the front (closest to the typist).  I suspect that worked well for the SOL enclosure it was aimed at but would suggest that for these standalone keyboards it be moved to the back.  Since the interface circuit board will presumably be sandwiched there, it makes sense to keep the front as thin as possible, adding thickness to the back where you might want the keyboard sloped a bit anyway.

dfnr2, 2022-01-01:
I think this issue is getting a bit far afield.  May I suggest opening up a thread on vcfed.org to discuss various aspects of the Knight/SAIL keyboards?

larsbrinkhoff, 2022-01-02:
Sure!  I opened up this thread:

https://forum.vcfed.org/index.php?threads/remaking-knight-and-sail-keyboards.1237093/

ams, 2024-09-11:
PCB for Space Cadet: https://github.com/rljacobson/SpaceCadet

One with Cherry MX switches: https://github.com/rigidus/SpaceCadet/tree/ModernCherryMx