osiweb/unified_retro_keyboard #40: Apple II: Spacebar return force and Reset key force to activate

Quantamm, 2021-12-31:
I'm assembling the Apple II version of the keyboard and I'm using Gateron red keyswitches (MX clones with a 45 cN activation force). I'm finding that the spacebar doesn't return all the way to the top of its travel on its own and it feels very mushy. Meanwhile, the reset key is as-easy as any other key to press (and therefore easy to accidentally activate).

The original Apple II keyboard solved both issues by having springs under those two keys.

I don't see anything in the Apple II build instructions about springs or increasing the force for these two keys. Am I missing something?

I'm thinking about putting Gateron blacks (60 cN) or greens (80 cN) under those keys. Is there a better way to increase the force needed?

dfnr2, 2021-12-31:
This isn't really an issue, so I will be closing it.  However, I do have the following comments:

1) This project doesn't specify what keys you use.  You can use futaba MD4-PCS, or any flavor of Cherry MX or its clones.  

2) If you are having trouble with the spacebar, most likely it is an issue with the stabilizers.  The stabilizers should work with any flavor of keys.  I have tested them with Cherry MX red, and the spacebar feels the same as other keys and returns properly.

3) The Apple II keymaps use CTRL-RESET for the RESET function, and SHIFT-RESET for the CLEAR_SCREEN function (for Apple I interfaces).  Simply hitting RESET will do nothing, so no extra spring needed.  That said, you can modify the keymap to perform a RESET on an unmodified RESET press if you want.  You can also choose whatever switch you want for the RESET key, or add a spring if you like.  Springs are available from multiple vendors.  That's up to you.

4) I tend to prefer Cherry MX/Gateron Black over the other Cherry MX variants.  Black has a little more force and feels closer to vintage keyboards.  Blue is not bad, but I'm not a huge fan of clicky.  I think that Red is more for gaming.

5) For posts discussing keyboard builds, questions, problems that aren't an issue with one or more of the objects in this repository or wiki, consider opening a thread on the vcfed.org or applefritter forums.  I will consider opening up a unified keyboard project forum for general discussions.

Quantamm, 2022-01-01:
I think you misunderstood my intent. I opened this issue because I'm following the instructions on the wiki to the letter and the end result isn't working, which means that there's a documentation gap and I was hoping that we could identify what it was so we could fix it. Closing my issue without doing anything to resolve it isn't going to help the next person.

dfnr2, 2022-05-26:
@Quantamm I see what you are saying.  The RESET key not being stiff really won't be much of an issue due to the two-key combo, but some folks might want the original behavior, and if that is added as a keymap, then the spring would be helpful.  Also, while I haven't seen an issue with the spacebar, the red keys are lower force, so I could see this becoming an issue.  

If you have tried out any springs and can recommend some, I will try out the proposed solution, and add it in to the BOM and instructions.

Quantamm, 2022-05-29:
@dfnr2, thanks for getting back to me. Here's what I ended up doing:

For everything except the reset and spacebar keys, I used Gateron Red switches.

For the reset key, I went with a Gateron Green switch. The heavier activation force and the audible click gave me the behavior and feel that I was looking for.

For the spacebar, the pre-bent stabilizer rod that I received was ever-so-slightly too short to the point where it slightly rubbed up against the sides of the plastic stabilizer guides. It wasn't noticeable when operated by hand, but was a problem once the whole board was assembled. If anything, I think that if https://github.com/osiweb/unified_retro_keyboard/wiki/apple-ii-notes , under step 2 of Prepare the keyboard PCB, had me friction fit the key switch and install the spacebar keycap, I probably could have caught it in time. As it was, I didn't find out until after all of the key switches were soldered in place.

I ended up buying a pack of 1/16" x 12" brass rods (https://www.amazon.com/gp/product/B009YO48LK), 3D printing a jig, and bending a new stabilizer rod. I then applied switch lube (https://www.amazon.com/gp/product/B096MHPHBM) to the plastic stabilizer guides. I used a Gateron Black switch and in the final assembly, I added two 9/32" x 1/2" compression springs (https://www.amazon.com/gp/product/B079P5P9LL) over each of the key switch posts. This got me to about 90% of the feel that I was looking for.

Like I mentioned above, I don't think all of this would have been necessary had I thought to check the spacebar behavior with both the key switch and key cap in place before soldering. I could have probably made some minor tweaks at that point, but once it was assembled, my options got limited.

dfnr2, 2022-05-30:
@Quantamm Good detail.  I will add a step to check the stabilizer fit early in the assembly as you suggest.

Perhaps it will be good to add a section about tuning the spacebar.  

I'm curious if you checked to see if the metal stabilizer bar was also a little tight on the aligner plate, or if it fit properly.  The stabilizer can also be checked using the mounting posts between the "X" and "." keys.

Are you willing to share your 3D model for the bending jig?  I have a jig made from a couple of wire-bending tools mounted to a piece of plywood, but it's a bit on the expensive side, and tricky to assemble perfectly.  A 3D-printed jig would be more reproducible and accessible.

Did you use the compression springs for all keys, or just the keyboard?  For the keyboard, did you use just one spring fot the key switch, or three--one on the switch and two on the stabilizers?

Quantamm, 2022-06-05:
> I'm curious if you checked to see if the metal stabilizer bar was also a little tight on the aligner plate, or if it fit properly. The stabilizer can also be checked using the mounting posts between the "X" and "." keys.

I did check it, but I don't remember what the result was. If it was way off, I would have stopped, but the bar was only a little short. I can't remember if I noticed if it fit well, or barely fit.

> Are you willing to share your 3D model for the bending jig? I have a jig made from a couple of wire-bending tools mounted to a piece of plywood, but it's a bit on the expensive side, and tricky to assemble perfectly. A 3D-printed jig would be more reproducible and accessible.

I checked my files and I no longer have it. It wasn't anything complicated. It was just a box with a height of whatever the lengths of the arms of the bar needed to be (~8mm) with two holes separated by whatever the length of the bar is (~120mm). I do recall having to print it a couple of times to get the finer measurements just right.

> Did you use the compression springs for all keys, or just the keyboard? For the keyboard, did you use just one spring fot the key switch, or three--one on the switch and two on the stabilizers?

I used springs only over the two stabilizer posts of the spacebar. The keyswitch does not have a spring.